Fiberprinter/fiberprinter-electronics #3: beschleunigungssensor

JanHTL, 2024-06-08:
beschleunigungssensor auswwählen und in projekt einbinden. Anschließbar machen

GodGotzi, 2024-06-08:
adxl335 find ich gut dafür gibt es fertige entwickler boards, gibt aber noch günstigere Alternativen

GodGotzi, 2024-06-08:
adxl335 adafruit

GodGotzi, 2024-06-08:
https://www.amazon.de/Beschleunigungssensor-Accelerometer-Digitales-Schwerkraft-%C3%9Cbertragung/dp/B0CMHTQF68/ref=sr_1_9?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dib=eyJ2IjoiMSJ9.qNW4VVCJh9KLxR3GBqggu4MqG-N4S36SM2oYTwf89uoEuP4_x4tbS_D08HvDHsVqPU_d1haVYUsnwqNq1bI4ZlyNHPGZpPYeSGuK-AhX7Y820chiDLeVnT726CxpUMIDJ8slUGq7tyZjqNfdxbdWbRbxNksqQOYc5tyvvwV7SzuQyKlPQWjfpAS_REKUxEiGcSa4Ql-0C2uYIF08CdL2vAQHaKFItEFa4V_00Vy4NmOtYE2i-fBGZFVkMS1NDgI2PJKf1C18VG9Dcxa54S3RhLGIU0Oc8D53hfu0vLnvqVs.F9JTdnXiz8mqwFT6OiskKnTccid-lK5x2B1Ad_7XkIM&dib_tag=se&keywords=beschleunigungssensor&qid=1717885949&sr=8-9

JanHTL, 2024-06-08:
sieht gut aus für den anfang werd ma den mal verbauen. von den anschlüssen her sind sie grötenteils sehr ähnlich

closed 2024-06-08 by JanHTL