Fiberprinter/fiberprinter-electronics #4: frage zu on off fans

JanHTL, 2024-06-08:
kann man diese direkt vom LED Driver zur Schaltung schliessen, wo bei den anderen das pwm signal kommt?

GodGotzi, 2024-06-08:
wie meinst du das?

GodGotzi, 2024-06-08:
wir brauchen fix nen mosfet oder transistor zum schalten. Wenn ich mich net ganz täusche hat der led driver pro pin max 30ma. Bedeutet wir brauchen ganz sicher noch einen Transistor oder Mosfet.

JanHTL, 2024-06-08:
> wir brauchen fix nen mosfet oder transistor zum schalten. Wenn ich mich net ganz täusche hat der led driver pro pin max 30ma. Bedeutet wir brauchen ganz sicher noch einen Transistor oder Mosfet.

ja fix
können wir den transistor nehmen tho den wir eh schon verwenden für die anderen die wir mit pwm antreiben

GodGotzi, 2024-06-08:
ja könnma

JanHTL, 2024-06-08:
> ja könnma

supi supi dann kann is so lassen wies is

closed 2024-06-08 by JanHTL